franciscofranco/FK-Kernel-Manager-IssuesAndFeatureRequest #1: Kernel backup compression

kdrag0n, 2019-02-27:
Currently, a kernel backup is uncompressed. Each backup takes as much space as the boot partition, though usually only up to ~30 MB of the partition is used.

Compressing backups would be desirable to conserve space and potentially even *reduce* backup time as there is less I/O necessary. Zstandard (or Zstd for short) is a modern compression algorithm that is fast and compresses fairly well, not too far from deflate in terms of compression ratio while being much faster. The official file extension suffix for Zstd-compressed files is `.zst`. I would recommend using this library: https://github.com/luben/zstd-jni

Another thing related to backup file extensions: you should name the files with `.img` as they are raw boot images, rather than not including any file extension. For example, after compression is added, the name of a backup could be `StockFeb2019.img.zst` on disk and `StockFeb2019` in the UI.

kdrag0n, 2019-02-27:
In testing, the size of the boot image dropped by 60% after being compressed with the default settings of Zstd, from 64 MB down to 25.84 MB.

franciscofranco, 2019-03-19:
I'm not going to bundle anything that requires JNI to operate. I'll investigate simple compression for version 4.1.

kdrag0n, 2019-03-19:
@franciscofranco There are pure Java implementations of Zstd available.

franciscofranco, 2019-03-19:
Then I will investigate 👍