franciscofranco/FK-Kernel-Manager-IssuesAndFeatureRequest #18: Governor profile

Remicaa, 2019-03-23:
The governor profile doesn't seem to work.

[image]
Steps:
1. Tap on "create a new profile"
2. Enter profile name
3. Tap "ok"

Nothing happened

franciscofranco, 2019-03-23:
Hmm. I just tried with all my devices and it worked fine. Which device you're on? What's the app version you're on?

Remicaa, 2019-03-23:
App version : 4.0.2 (1903221853)
Device : mi max 2 
Rom : Miui 9.3.21
Android version : 7.1.1

Or am I missing a step to create one?

franciscofranco, 2019-03-25:
Can you check if the profile is in /sdcard/franco.kernel_updater/governor_profiles/ after you save it?

Remicaa, 2019-03-26:
It's not there, but if I put a blank file there, its showed up in the app

Remicaa, 2019-03-26:
Sorry miss clicked

franciscofranco, 2019-04-11:
I really can't reproduce. I made some improvements for robustness for version 4.1. Re-open this if it's still "broken" after I release 4.1.

closed 2019-04-11 by franciscofranco