franciscofranco/FK-Kernel-Manager-IssuesAndFeatureRequest #33: Inconsistency in LMK presets

groozchique, 2019-03-30:
By default, values for LMK settings are in *Megabytes*, but when you're trying to change it via dialog window – numbers are different, so it's not that convenient as it meant to be.

Also as UX improvement, you can add arrows to LMK values like you did in *Display control*, so it's easier to tune them precisely.

[image]
[image]

franciscofranco, 2019-03-31:
Agreed, will fix for the next release.

franciscofranco, 2019-04-11:
Fixed for 4.1.

closed 2019-04-11 by franciscofranco