franciscofranco/FK-Kernel-Manager-IssuesAndFeatureRequest #36: Klapse settings

gansaid, 2019-04-03:
Device: OnePlus 5T
ROM: OxygenOS Latest 
Kernel: FK-r44

Klapse settings (start and end times, durations, red blue and green levels) reset after every reboot

franciscofranco, 2019-04-10:
I can't reproduce on all my +20 devices. Every setting is properly re-applied on boot. That said, I can make improvements to the boot service and make it more reliable.

franciscofranco, 2019-04-18:
Improvements added for 4.1.

closed 2019-04-18 by franciscofranco

kocane, 2019-05-19:
@franciscofranco  using latest version and latest kernel on my OP6, I experience this as well. On reboot, k-lapse is disabled. I've tried toggling things some times and I've tried setting "set on boot" both before and after I configure k-lapse but it doesn't help.