franciscofranco/FK-Kernel-Manager-IssuesAndFeatureRequest #37: kernel restore fails

jancm, 2019-04-06:
Trying to restore a saved kernel backup appears to work, but after a reboot the original kernel is still being used. Pixel 3 XL

Let me know if you need a log, or anything else.

franciscofranco, 2019-04-10:
I don't have a Pissel 3 XL to test, but the restore is working fine for me on all my devices. It reads the actual active partition and then writes the backup to it.

Can you give me a bit more information on your entire procedure, step by step as detailed as possible. Thanks.

jancm, 2019-04-11:
Nothing super complicated. Just tried it again. My current kernel is Kirisakura 2.2.0
I have (among others) a Magisk patched stock boot image so I tried to restore to that. Fired up FKM, went to kernel backups section. Clicked on the saved backup image. Got asked  if I want to restore this kernel. In this case chose flash only since I was recording logcat. Flash finished. I rebooted. Checked running kernel. It was still Kirisakura. Unfiltered logcat attached, in case it'll help.

[2019-04-11-05-22-27.txt](https://github.com/franciscofranco/FK-Kernel-Manager-IssuesAndFeatureRequest/files/3067972/2019-04-11-05-22-27.txt)

jancm, 2019-04-11:
Further investigation, based on hashes, it looks like what's happening is that the current backup is getting overwritten by the running kernel. In other words the copy is going in the wrong direction.

franciscofranco, 2019-04-12:
Just found the issue. Will be fixed for 4.1. Thanks & apologies.

closed 2019-04-12 by franciscofranco