franciscofranco/FK-Kernel-Manager-IssuesAndFeatureRequest #41: Performance Profiles Unsupported

DenyDarko, 2019-04-10:
Hey @franciscofranco :)

I'm having a kinda weird issue on my OP6. 

Performance Profiles are unavailable while I'm on r15.

Attaching some screenshots and a logcat. 

[image]
[image]
[logcat_04-11-2019_00-32-19.txt](https://github.com/franciscofranco/FK-Kernel-Manager-IssuesAndFeatureRequest/files/3065957/logcat_04-11-2019_00-32-19.txt)

franciscofranco, 2019-04-10:
Hi. You need to flash the Kernel AFTER flashing magisk.

closed 2019-04-11 by franciscofranco

DenyDarko, 2019-04-11:
Well that's how I did it and it worked at the beginning xD

I guess I need to reflash the kernel after every Magisk update as I just reflashed kernel and it worked. 

How about a check in Kernel Manager about that and a warning if kernel needs reinstall? 😁