franciscofranco/FK-Kernel-Manager-IssuesAndFeatureRequest #50: GPU Maximum frequency and GPU Governor can not changed

Alexey71, 2019-04-18:
GPU Maximum frequency and GPU Governor can not changed

https://drive.google.com/file/d/1IhlYBkAgvp806wTlT8ee3lM8mCvSsR12/view?usp=sharing

_Originally posted by @Alexey71 in https://github.com/franciscofranco/FK-Kernel-Manager-IssuesAndFeatureRequest/issues/49#issuecomment-484374042_

franciscofranco, 2019-04-21:
The governor is not suposed to be changed to anything other than performance. The device needs the msm-adreno-tz to be running correctly. That's not my app's fault, it's a protection from the kernel.
About the max freq, it's working as intended. I don't know where Kernel Auditor is getting the freqs from, but I changed on my app and checked with the console and changed just fine after changing it to 400 MHz on my app:

mido:/ # cat /sys/class/devfreq/1c00000.qcom,kgsl-3d0/max_freq
400000000

There is no bug since my app is working as intended, so I'm closing this.

closed 2019-04-21 by franciscofranco

Alexey71, 2019-04-21:
/sys/devices/soc/1c00000.qcom,kgsl-3d0/kgsl/kgsl-3d0/max_gpuclk

franciscofranco, 2019-04-22:
That's not the correct path...

franciscofranco, 2019-04-22:
/sys/class/devfreq/1c00000.qcom,kgsl-3d0/max_freq