franciscofranco/FK-Kernel-Manager-IssuesAndFeatureRequest #62: White mode is automatically being applied.

paschfire, 2019-05-18:
[image]
[image]

Here to report my eyes are broken! 😵

Oh ya... You also cannot disable "White mode"

If you choose Dark and then exit/enter FKM, when relaunched the eye bleeding white is applied again. With Dark showing enabled.

#WhiteUIMustDie

paschfire, 2019-05-18:
Clearing FKM App data seems to have solved the issue for now, Dark mode is remaining applied upon exit/enter FKM.

So might just be a single case issue, if so pls close this ticket.

franciscofranco, 2019-05-18:
Fixed.

closed 2019-05-18 by franciscofranco