franciscofranco/FK-Kernel-Manager-IssuesAndFeatureRequest #63: Color Calibration Problems

NeroOrin, 2019-05-21:
Changing the values of the sliders for Saturation, Contrast, Value, and Hue don't do anything. RGB works fine though.

franciscofranco, 2019-05-21:
It works fine for all my devices. A little more context & information about your device otherwise I can't help. Have you tried screening off/on ? Some devices need that.

NeroOrin, 2019-05-24:
Even when turning the screen on/off the settings do nothing. It sometimes works but I don't know when. When it does work, whenever I rebootn the device, all the settings wont take effect anymore. I've tried re-installing the manager and re-flashing the kernel.

franciscofranco, 2019-06-13:
And which device is that? I really can't reproduce no matter what.

closed 2019-07-09 by franciscofranco

ibenk76, 2019-08-29:
Color calibration not work in pocophone

franciscofranco, 2019-08-29:
Yes it does. Turn the display off/on.