franciscofranco/FK-Kernel-Manager-IssuesAndFeatureRequest #68: Battery Monitor Service

whoknowswhoiam, 2019-05-23:
Device: OnePlus 5
ROM: OxygenOS
Kernel: Weeb Kernel unreleased build
Root status: Unrooted
Issues:
1. Rebooting the device disables the Battery Manager Service automatically and requires it to be re-enabled by the user again.

While doing this, I was also able to reproduce the battery stats not resetting properly. On the first reboot where I flashed a new kernel from TWRP, I had 1.5 hours of SoT with 1% dropped, implying the previous boot time's SoT was still being accounted for, which it shouldn't since I have that option disabled.
This issue was first brought to your attention here:
https://github.com/franciscofranco/FK-Kernel-Manager-IssuesAndFeatureRequest/issues/28

2. There is no option to get to settings from the Battery Manager Service menu. This means, in order to change a setting, the user would have to root. However, I understand that most of the options in the settings menu are useless when unrooted but I thought it was worth mentioning regardless.

baalajimaestro, 2019-05-23:
> " This means, in order to change a setting, the user would have to root"

I see no point of a kernel manager without root.

whoknowswhoiam, 2019-05-23:
> > " This means, in order to change a setting, the user would have to root"
> 
> I see no point of a kernel manager without root.

A bug is a bug nonetheless.

franciscofranco, 2019-06-13:
I'll make this a bit easier for 4.3.

franciscofranco, 2019-07-01:
Implemented for 4.3.

closed 2019-07-01 by franciscofranco