franciscofranco/FK-Kernel-Manager-IssuesAndFeatureRequest #78: Modifying CAF CPU Boost frequency selection behaviour

buzzcut-1105, 2019-06-22:
Currently, changing frequencies for both/two clusters is a little cumbersome i.e. : 

1. Select the 'Input boost frequency' option 
2. Select new Little cluster frequency 
3. Select the 'Input boost frequency' option 
4. Select new Big cluster frequency 

There is a sudden/jarring transition after selecting a frequency, i.e. between steps 2 and 3, where it closes the selection window, after which you have to return to the frequency selection window again. 

What I'd like to suggest is instead of having to repeat Step 1 twice, make it so : 

1. Select the 'Input boost frequency' option
2. Select frequencies for both clusters (without having the window exit)
3. (Maybe) Press a 'Done' button, or something like that?

In my opinion, adding an explicit 'Done' or 'Update' option would result in a _slightly_ better UX. Would love to hear your thoughts on it!

franciscofranco, 2019-06-26:
Agreed - will implement. Thanks for your suggestion.

franciscofranco, 2019-06-28:
Done. Will be released on 4.3.

closed 2019-06-28 by franciscofranco