franciscofranco/FK-Kernel-Manager-IssuesAndFeatureRequest #84: Duplicate CPU stats

Remicaa, 2019-07-06:
[image]

It shows the cpu stats twice when first time opening live monitor, i have cpu stats as default page

Remicaa, 2019-07-06:
It doesn't happen before 4.3 update,  soc is msm8953

Remicaa, 2019-07-09:
Still persist in 4.3.1

franciscofranco, 2019-07-09:
Hmm having a hard time reproducing it.

Remicaa, 2019-07-10:
Sorry i forget to mention this, i set the default "System Health" to cpu frequency stats, it doesn't happen if i set to the other tab , but doesn't seems to happened on stock kernel. Probably a derp in my kernel, you can close this issue if you want

Remicaa, 2019-07-14:
Fixed in 4.3.3 update , thanks

closed 2019-07-20 by franciscofranco

dhruvan2912, 2019-08-16:
I'm facing this issue in 4.4.1 
[image]
[image]

franciscofranco, 2019-09-01:
Fixed for the upcoming 4.6.