franciscofranco/FK-Kernel-Manager-IssuesAndFeatureRequest #94: Big little cluster detection (?)

zputnyq, 2019-07-17:
Hi @franciscofranco 

I'm new user of FK kernel manager.
It's confusing that I found out about big little cluster detection is different to Kernel Adiutor, on FK, it's showing the opposite of KA. I don't know whether it's a bug or not. I hope you don't mind looking at it.
I'm on xperia m4 aqua (SD 615) stock rom 6.0.1 & stock kernel.

Thanks, 
I really appreciate it

franciscofranco, 2019-07-20:
I have absolutely no idea how that's even possible :/

gandalf02, 2019-07-22:
Yes sir , it's possible , cause some Snapdragon chipsets have 0-3 as little CPUs , and 4-7 as big unlike the generic ones ,like sdm435 . May be kernel auditor has a different algo for detecting the big and little clusters. Or may be u implemented big and little core algo by just taking 0-3 as big and 4-7 as little like all other socs like 8996 have.

gandalf02, 2019-07-22:
> I have absolutely no idea how that's even possible :/

Yes sir , it's possible , because some Snapdragon chipsets have 0-3 as big CPUs , and 4-7 as little like sdm435 . May be kernel auditor has a different algo for detecting the big and little clusters. Or may be u implemented big and little core detection algo by just taking 0-3 as little and 4-7 as big(for 4 cores 0-1 are little and 2-3 big ) like all other socs like 8996 have.

franciscofranco, 2019-07-24:
@kdrag0n gave me a good idea to tackle this. I'll ship it for 4.4.1 or something, currently no time to work it out for 4.4.

franciscofranco, 2019-09-02:
Should be working fine now. Re-open if not.

zputnyq, 2019-09-06:
Still the same on 4.5.1 & 4.6
[image]
[image]

franciscofranco, 2019-09-06:
Shit :)

franciscofranco, 2019-11-14:
It's been fixed.

closed 2019-11-14 by franciscofranco