franciscofranco/FK-Kernel-Manager-IssuesAndFeatureRequest #97: GPU boost value cannot be changed, FKM PP QS tile(Balanced) has possible visual issue.

paschfire, 2019-07-21:
FKM v4.3.4

GPU boost can no longer be adjusted, I had it previous set to low but taping on it now does nothing but highlight the option, no overflow menu is shown.

The PP QS tile used to be highlighted using the current system colours, now when set to Balanced it's just grey(inactive). Performance and Battery are still highlighted. Just not sure if this was an intended change made within FKM or not. Screenshots attached to show examples.

OP3T, OOS 9.0.4, FK r48, Magisk 19.4

franciscofranco, 2019-07-23:
Gpu boost derp is not expected and will fix, the other is intended since Balance profile is the default so it shouldn't be shown as "Active". :)

franciscofranco, 2019-07-24:
Gpu boost fixed for 4.4.

closed 2019-07-24 by franciscofranco