franciscofranco/FK-Kernel-Manager-IssuesAndFeatureRequest #101: Q - missing old kernel backups

jancm, 2019-08-29:
Moved to Q beta 6. I had a bunch of kernels saved, some of which are relevant to Q 

FKM isn't displaying any of the previous kernels. A new backup, which stores in the same directory, does appear.

Also noticed that FKM is no longer generating an md5 file when creating a backup. 

I'd be happy to provide logs, or anything else, just let me know what you need.

Pixel 3 XL, latest Canary Magisk, latest FKM (4.5.1)

franciscofranco, 2019-09-01:
Hmm. I've made no changes on that part of the code and I don't delete stuff from the sdcard without user consent so something was fucked when you moved to Q since the backups should show as they're stored in the sdcard... I'll try to reproduce on my pixel 2xl

jancm, 2019-09-01:
There's nothing actually getting deleted. It's just not seeing what's
there. If you want logs just let me know what exactly you want.

Other quirks. No log is being produced when I flash Magisk modules. Log is
fine for kernel flash. As mentioned - no md5 produced for saved kernel
image. Also the previously saved image, that initially appeared, no longer
does after a reboot (I do need to confirm that the reboot is what caused
this)

This all started after I switched to Q beta 6. It was working fine on Pie
so the code isn't bad across the board, it's a Q issue of some kind (I
think). If it has anything to do with system-as-root (although I can't
imagine it does) then you probably won't be able to duplicate on the 2.

franciscofranco, 2019-09-20:
Should be fixed on 4.6.2. Fucking Scooped Storage shite from Google.

closed 2019-09-20 by franciscofranco