franciscofranco/FK-Kernel-Manager-IssuesAndFeatureRequest #107: zRam compression problems

jancm, 2019-09-20:
zRam compression still isn't getting correctly applied. Pixel 3 XL, Android 10
Defaults at the moment are:
enabled
2048
lz4
100 (swappiness)

I want to change this to zstd and 175.
I set the compression algorithm to zstd, set swappiness to 175. Turn on apply at boot, just for the 2 changed items. Takes a while but it makes the requested changes.

Reboot the phone.

zRam starts up correctly. Enabled, 2048, lz4, 175.
So the only thing that needs to change is the algorithm, everything else is as it should be. 

It next goes into a 'zRam is shutting down' phase. I assume this is preliminary to changing the algorithm. However it never starts up again.

Go to check after a few minutes zRam is disabled. Enable it and it comes back with 512MB rather than 2048. And the compression algorithm is still lz4

franciscofranco, 2019-11-02:
Yeah ZRAM is not easy to nail because the whole process of resizing/disabling/changing agorithms is pretty heavy and demanding and it gets exponentially worse depending on the ZRAM size/usage.
I'll run through this a couple more times to see if/where is the issue with the set on boot. It runs the same code as when you toggle it through the app so I don't know. I'll check.

jancm, 2020-05-15:
This has been working correctly for quite some time now (at least for me). This issue should probably be closed.

closed 2020-05-19 by franciscofranco