franciscofranco/FK-Kernel-Manager-IssuesAndFeatureRequest #108: Big Cluster "Sampling" on N6P running Android 10

armyav8tor, 2019-09-24:
I'm running PixelExperience 10 on my Nexus 6P and FK Kernel Manager reports big cluster as Sampling and never shows the Frequency.  Under Live Monitor, it shows 0% for all CPUs and under CPU STATS, the big cluster seems to be missing a lot of data.  Everything reports fine under EX but I like the FK interface more.  Please let me know if there's something we can do to get it to report correctly, thanks!!

[image]

[image]

[image]

franciscofranco, 2019-11-02:
Hmm, that happens if the cores are offline. As you see the big cluster freq reports as "offline". Though I see your point, it shouldn't say "Sampling" in there. I'll fire up my N6P and run some tests. Don't expect a fix for the next small release since this requires some time for debugging, but as soon as I figure this out I'll push the fix to the store.

franciscofranco, 2019-11-07:
Fixed for next release.

closed 2019-11-07 by franciscofranco