franciscofranco/FK-Kernel-Manager-IssuesAndFeatureRequest #110: ZRam resizing and enabling or disabling issue.

jignesh071, 2019-10-02:
ZRam disbale takes to many tries. It automatically got enable next time you open the app. Also Resizing Zram having same issue.

franciscofranco, 2019-11-02:
The ZRAM resize simply takes time depending on the size of the ZRAM block/usage. So once you ran the "resize" option you simply have to wait until the system re-allocates everything and resizes the block. Same thing as disable.

closed 2019-11-02 by franciscofranco

jignesh071, 2019-11-09:
But it takes so much time to do that and once you exit or goes to another
app it goes back to where it was, so that's weird. Let it do that thing in
background so that you wouldn't have to do it again and again untill it
resizes or wait untill it resizes.

Thanks & Regards
Jignesh Panchal
Palanpur, Gujarat, India
☎️ +919898704540