franciscofranco/FK-Kernel-Manager-IssuesAndFeatureRequest #115: Incorrect readings in battery monitor service on rmx1851

narshishukla, 2019-11-06:
I just got a new device and fk kernel manager is showing incorrect current input, its always showing 1mA and it have issues refreshing battery percentage and temperature, i have checked background process settings and its fine there. 
Same issue on other realme devices that i checked i.e realme xt and 5pro
Running color os and everything stock on realme 3 pro

franciscofranco, 2019-11-07:
Hi. Well I just read the values the framework sends me. Looks like yet another OEM fuckery. I don't have those devices so I don't know how I can add a workaround..

narshishukla, 2019-11-09:
Hey dev,
Thanks for your time to reply my mail.
Anyways, Can i help you in any way? Like uploading apks or providing logs?