franciscofranco/FK-Kernel-Manager-IssuesAndFeatureRequest #126: Display Pixel density

diegomenfer, 2020-02-23:
When editing a profile that already exists, the text box for editing Display Pixel Density value, don't show up, instead it shows options for Display Orientation
[image]

FKM 5.1.1
Asus Max Pro M1 (X00T)
Lineage 16 Official

lambofgodmtldev, 2020-12-28:
Having this issue too, frustrating as I'm on a custom variable DPI with tasker (along with resolution) and every time an app triggers the profile the DPI changes.

Is there a way to leave the density unchanged?

Liamb135, 2021-01-11:
Also having the same issue. When reverting from a custom profile FKM sets my DPI to a low value.