franciscofranco/FK-Kernel-Manager-IssuesAndFeatureRequest #128: "Franco Kernel Manager keeps stopping" when entering Klapse settings

Iron-E, 2020-05-15:
# Context

I use the Franco Kernel and the Franco Kernel Manager on my LineageOS 16.0 (`cheeseburger`/OP5).

I am rooted with Magisk v20.4.

I use Franco Kernel Manager primarily to adjust Klapse settings so that the "temperature" of my phone's display adjusts as the day passes on to become warmer at night.

# Description

Before FKM version 6.0, I had been having difficulties with #36. However, since 6.0 released, I have been unable to go into the `Display control > Tools > Klapse` menu at all.

Each time I attempt to enter this menu, the app crashes. After two consecutive attempts, I receive the standard "`XYZ App` keeps stoping" mesasge.

I can't downgrade because Google Play doesn't provide previous versions of apps.

# Steps to Reproduce

1. Open Franco Kernel Manager.
2. Go to the `Display control` menu.
3. Navigate to `Tools > Klapse` and enter the menu.
4. Observe that the app has crashed.

# Expected Behavior

* Franco Kernel Manager will allow me to adjust Klapse settings.

# Logs

Let me know what level of log you'd like and I can grab them for you with `logcat`.

jancm, 2020-05-15:
I'm having the same issue. Pixel 3 XL. May update. Kirisakura kernel.
Logs attached.
[2020-05-15_06.03.zip](https://github.com/franciscofranco/FK-Kernel-Manager-IssuesAndFeatureRequest/files/4636732/2020-05-15_06.03.zip)

jancm, 2020-06-08:
FKM 6.1
Still happening

Iron-E, 2020-07-07:
I would offer to check this out, but I don't know the first thing about Android development. @franciscofranco will this be checked out, or is this `wontfix`? I know there have been issues similar to this before that haven't been reproducible on your side.

franciscofranco, 2020-07-07:
I'll take a look asap. Apologies

Iron-E, 2020-10-05:
@jancm are you still getting this?