franciscofranco/FK-Kernel-Manager-IssuesAndFeatureRequest #159: What does GPU pwrlevel do?

fusk-l, 2022-08-03:
I have a default value of 4, what does it do if i set it to either 3 or 5? Can't find any explanation anywhere.
If i change the value, when i return to the settings it's reverted to 4, why is that?

Sawminye114, 2023-10-07:
[https://github.com/franciscofranco/FK-Kernel-Manager-IssuesAndFeatureRequest/issues/159#issue-1327678996](h)

Peppe289, 2023-12-17:
> I have a default value of 4, what does it do if i set it to either 3 or 5? Can't find any explanation anywhere. If i change the value, when i return to the settings it's reverted to 4, why is that?

The GPU pwrlevel indicates the frequency level.
Everything as the default is defined in the kernel [1]. The maximum value indicates that it starts from the lowest level, while the level 1 or 0 indicates the highest ones. As you can see for example for sdm660 it is defined in the appropriate dtsi with the name of `qcom,gpu-pwrlevel@X`

On qcom devices this depends on how the value is managed.
 
In the kernel, to read the user entry points [2], a procedure is used that captures the value entered, verifies its correctness and carries out the operations. It is a synchronous procedure. The same thing happens for writing (in the `DEVICE_ATTR` you can see the name, permissions, addresses of the procedures for reading (show) and writing (store)). As soon as something is changed it is reported in the entry pointer. According to your problem it seems that the kernel developer has blocked the input procedure, making the value change ineffective.

[1] https://github.com/android-linux-stable/msm-4.4/blob/kernel.lnx.4.4.r40-rel/arch/arm/boot/dts/qcom/sdm660-gpu.dtsi#L70
[2] https://github.com/android-linux-stable/msm-4.4/blob/7ce3eeca3360577db5284c3ff30a63e26adddcfd/drivers/gpu/msm/kgsl_pwrctrl.c#L1473

I hope this explanation was useful to you. Best regards.