franciscofranco/FK-Kernel-Manager-IssuesAndFeatureRequest #161: Per-app profile issue

MALWAREKILLS, 2022-08-29:
when creating a new per-app profile and launching the applied apps
it works at the first time after closing and opening the applied apps it isn't working
note: clearing the FKM app data and re-creating the per-app profile works again for once

jaswinprakash, 2022-08-29:
I have the same issue , also i found these error in accessibility section 
[image]
[image]

MALWAREKILLS, 2022-08-29:
mine even the accessibility option stays ON but
still it won't work
just 1 time after clear data
[image]

closed 2023-07-03 by MALWAREKILLS