franciscofranco/FK-Kernel-Manager-IssuesAndFeatureRequest #167: log gets flooded by: background should not be translucent

laviniuc, 2023-01-29:
a13, after about 24h

norrowed it down to the batt monitor service persistent notification

[image]

it works ok for another day or so after restarts, but once it starts logging the only way to stop it was disabling notifs for the whole fkm or disabling the batt mon service...

lemonderon, 2024-01-26:
I'm experiencing the same issue on CrDroid 9.13 on 1+6. Not running FK. Also seen this issue reported on https://community.spotify.com/t5/Android/Spotify-Android-app-is-causing-ContrastColorUtil-to-behave/td-p/5619455. The person there also runs CrDroid v9 A13. 
@laviniuc Could you give some details about your device+OS?

laviniuc, 2024-01-26:
can't remember the exact device name (probably mi 11?), it was a xiaomi for sure running crdroid 9 (no such logs on the previous cr8 based on a12)

it's probably related to the background settings of the notifications (somehow) in a13, you should check which apps have persistent notifications and turn them off one at a time (or all), that should help you find your culprit.

i remember it was a pain to troubleshoot/figure it out because it takes a long time for the problem to show up.

lemonderon, 2024-01-26:
Thank you. Will try to contact the CrDroid devs then.